canonical/firmware-updater #4: Suggestion: use an Elevated Button if there is an upgrade available

Feichtmeier, 2022-01-31:
[image]

I think it would be nice to forward the hasUpgrade bool to the buttons and then use an elevated button instead of the outlined button, WDYT @jpnurmi ?

jpnurmi, 2022-01-31:
I like the idea even though should it be in the dialog because there can be multiple releases available?

Feichtmeier, 2022-01-31:
The green pill suggests "nice you have an update here that's a good thing". Then you expand the tile and the two buttons are kind of hiding by their wording and subtle visuals that you can open the actual dialog right here. Maybe the wording can be changed to "manage upgrades" and inside the dialog the update tile is also somehow marked to be more engaging?

jpnurmi, 2022-01-31:
Yes, makes sense to draw the user's attention. Upgrading is the main use-case for this app after all. :)

By the way, there are probably more buttons/actions required. I think those are "utility" types of actions, though, similarly to the verify button. Perhaps they could be moved out of the way into a ⚙ popup menu button...

PR #11 (2022-01-31): Make updates more engaging

| Before | After |
| --- | --- |
|[image] |[image] |
|[image] | [image] |

Closes #4
closed 2022-01-31 by jpnurmi