canonical/firmware-updater #28: Snap: can't connect to fwupd

Feichtmeier, 2022-09-08:
when running

sudo snap connect firmware-updater:fwupd 

the command responds that the policy does not allow this action :(

jpnurmi, 2022-09-08:
@kenvandine
```sh
$ sudo snap connect firmware-updater:fwupd
error: cannot perform the following tasks:
- Connect firmware-updater:fwupd to snapd:fwupd (connection denied by slot rule of interface "fwupd")
```

fthx, 2022-09-24:
I guess this problem explains why firmware updater start with a void window?

```
~$ firmware-updater 
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.169" (uid=1000 pid=6701 comm="/snap/firmware-updater/10/bin/firmware-updater " label="snap.firmware-updater.firmware-updater (enforce)") interface="org.freedesktop.DBus.Properties" member="GetAll" error name="(unset)" requested_reply="0" destination="org.freedesktop.fwupd" (bus)
#0      DBusClient._callMethod (package:dbus/src/dbus_client.dart:1124)
<asynchronous suspension>
#1      DBusClient.callMethod (package:dbus/src/dbus_client.dart:608)
<asynchronous suspension>
#2      DBusRemoteObject.getAllProperties (package:dbus/src/dbus_remote_object.dart:142)
<asynchronous suspension>
#3      FwupdClient.connect (package:fwupd/src/fwupd_client.dart:133)
<asynchronous suspension>
#4      FwupdService.init (package:firmware_updater/fwupd_service.dart:43)
<asynchronous suspension>
#5      Future.wait.<anonymous closure> (dart:async/future.dart:522)
<asynchronous suspension>
```

Feichtmeier, 2022-11-18:
I think this requires a change to the fwupd snap to open slots

kenvandine, 2022-11-18:
Just requires a store assertion, which is pending https://forum.snapcraft.io/t/autoconnect-request-for-firmware-updater/32505

fthx, 2022-12-08:
Seems to be solved if I read the link above?
I can start this software now.
:-)

jpnurmi, 2022-12-08:
Indeed! :)

closed 2022-12-08 by jpnurmi