canonical/firmware-updater #43: New design

jpnurmi, 2022-09-12:
| | | | |
|---|---|---|---|
| [image] | [image] | [image] | [image] |
| [image] | [image] | [image] | [image] |

Screenshots from https://miro.com/app/board/uXjVO8s0uD4= (not public, sorry)

Feichtmeier, 2022-09-12:
Is the navigation like chronological steps or a normal master details view?

jpnurmi, 2022-09-12:
Master-detail with "nested" navigation in the details view to replace the current popup dialog.

Feichtmeier, 2022-09-12:
Nested navigation sounds nice. Could help with the software app to replace the Dialogs with routes

jpnurmi, 2022-09-16:
#48 took `YaruMasterDetailPage` into use

closed 2022-09-16 by jpnurmi