canonical/firmware-updater #53: Error handling

jpnurmi, 2022-09-16:
We need a nice way to present errors to the user when something goes wrong.

Here's the full list of fwupd error codes for reference:
```
 * @FWUPD_ERROR_INTERNAL:			Internal error
 * @FWUPD_ERROR_VERSION_NEWER:			Installed newer firmware version
 * @FWUPD_ERROR_VERSION_SAME:			Installed same firmware version
 * @FWUPD_ERROR_ALREADY_PENDING:		Already set to be installed offline
 * @FWUPD_ERROR_AUTH_FAILED:			Failed to get authentication
 * @FWUPD_ERROR_READ:				Failed to read from device
 * @FWUPD_ERROR_WRITE:				Failed to write to the device
 * @FWUPD_ERROR_INVALID_FILE:			Invalid file format
 * @FWUPD_ERROR_NOT_FOUND:			No matching device exists
 * @FWUPD_ERROR_NOTHING_TO_DO:			Nothing to do
 * @FWUPD_ERROR_NOT_SUPPORTED:			Action was not possible
 * @FWUPD_ERROR_SIGNATURE_INVALID:		Signature was invalid
 * @FWUPD_ERROR_AC_POWER_REQUIRED:		AC power was required
 * @FWUPD_ERROR_PERMISSION_DENIED:		Permission was denied
 * @FWUPD_ERROR_BROKEN_SYSTEM:			User has configured their system in a broken way
 * @FWUPD_ERROR_BATTERY_LEVEL_TOO_LOW:		The system battery level is too low
 * @FWUPD_ERROR_NEEDS_USER_ACTION:		User needs to do an action to complete the update
 * @FWUPD_ERROR_AUTH_EXPIRED:			Failed to get auth as credentials have expired
 ```
https://github.com/fwupd/fwupd/blob/main/libfwupd/fwupd-error.h

jpnurmi, 2022-09-29:
I have a game controller that needs to be in a special FW update mode.

<details><summary>dbus-monitor</summary>

```
signal time=1664460478.014297 sender=org.freedesktop.DBus -> destination=:1.929 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.929"
signal time=1664460478.014318 sender=org.freedesktop.DBus -> destination=:1.929 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.929"
method call time=1664460490.211520 sender=:1.887 -> destination=:1.111 serial=24 path=/; interface=org.freedesktop.fwupd; member=GetRemotes
method call time=1664460492.354552 sender=:1.887 -> destination=org.freedesktop.fwupd serial=25 path=/; interface=org.freedesktop.fwupd; member=Install
   string "3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627"
   file descriptor
   array [
      dict entry(
         string "reason"
         variant             string "user-action"
      )
      dict entry(
         string "allow-older"
         variant             boolean true
      )
      dict entry(
         string "allow-branch-switch"
         variant             boolean true
      )
   ]
signal time=1664460492.369965 sender=:1.111 -> destination=(null destination) serial=6458 path=/; interface=org.freedesktop.fwupd; member=DeviceChanged
   array [
      dict entry(
         string "DeviceId"
         variant             string "3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627"
      )
      dict entry(
         string "Guid"
         variant             array [
               string "043c7673-bd45-5e2e-b8b9-c22df2dfec2a"
               string "b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e"
               string "1bf5ac96-99ce-54f2-abf4-7eabe6802533"
               string "81becceb-f87b-5c66-b30a-b0958043d65d"
            ]
      )
      dict entry(
         string "Icon"
         variant             array [
               string "input-gaming"
            ]
      )
      dict entry(
         string "Name"
         variant             string "SN30 Pro"
      )
      dict entry(
         string "Vendor"
         variant             string "8BitDo"
      )
      dict entry(
         string "VendorId"
         variant             string "USB:0x2DC8"
      )
      dict entry(
         string "Flags"
         variant             uint64 34376515746
      )
      dict entry(
         string "Created"
         variant             uint64 1664460446
      )
      dict entry(
         string "Summary"
[... 725 lines cut ...]
      dict entry(
         string "UpdateError"
         variant             string "Device has been removed"
      )
      dict entry(
         string "UpdateMessage"
         variant             string "Unplug the controller, hold down L1 + R1 + START until the red LED on top of the controller is flashing and then reconnect the controller."
      )
      dict entry(
         string "UpdateImage"
         variant             string "https://github.com/fwupd/8bitdo-firmware/raw/master/screenshots/sn30-pro.png"
      )
      dict entry(
         string "UpdateState"
         variant             uint32 5
      )
      dict entry(
         string "VersionFormat"
         variant             uint32 3
      )
   ]
```
</details>

[image]

jpnurmi, 2022-09-29:
Same device, different error. This time it actually started installing but then failed.

<details><summary>dbus-monitor</summary>

```
signal time=1664460953.337635 sender=org.freedesktop.DBus -> destination=:1.939 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.939"
signal time=1664460953.337651 sender=org.freedesktop.DBus -> destination=:1.939 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.939"

method call time=1664460958.316972 sender=:1.887 -> destination=:1.111 serial=27 path=/; interface=org.freedesktop.fwupd; member=GetReleases
   string "3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627"
method call time=1664460972.246123 sender=:1.887 -> destination=:1.111 serial=28 path=/; interface=org.freedesktop.fwupd; member=GetRemotes
method call time=1664460973.472311 sender=:1.887 -> destination=org.freedesktop.fwupd serial=29 path=/; interface=org.freedesktop.fwupd; member=Install
   string "3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627"
   file descriptor
   array [
      dict entry(
         string "reason"
         variant             string "user-action"
      )
      dict entry(
         string "allow-older"
         variant             boolean true
      )
      dict entry(
         string "allow-branch-switch"
         variant             boolean true
      )
   ]
signal time=1664460999.245460 sender=:1.111 -> destination=(null destination) serial=6591 path=/; interface=org.freedesktop.fwupd; member=DeviceChanged
   array [
      dict entry(
         string "DeviceId"
         variant             string "3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627"
      )
      dict entry(
         string "Guid"
         variant             array [
               string "78acf201-a624-5c0b-a521-e204be14e3d5"
               string "b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e"
            ]
      )
      dict entry(
         string "Icon"
         variant             array [
               string "input-gaming"
            ]
      )
      dict entry(
         string "Name"
         variant             string ""
      )
      dict entry(
         string "Vendor"
         variant             string "8BitDo"
      )
      dict entry(
         string "VendorId"
         variant             string "USB:0x2DC8"
      )
      dict entry(
         string "Flags"
         variant             uint64 34359746722
      )
      dict entry(
         string "Created"
         variant             uint64 1664460942
      )
      dict entry(
[... 559 lines cut ...]
      dict entry(
         string "VersionRaw"
         variant             uint64 133
      )
      dict entry(
         string "InstallDuration"
         variant             uint32 120
      )
      dict entry(
         string "UpdateError"
         variant             string "Device has been removed"
      )
      dict entry(
         string "UpdateState"
         variant             uint32 3
      )
      dict entry(
         string "VersionFormat"
         variant             uint32 3
      )
   ]
signal time=1664461002.306883 sender=:1.111 -> destination=(null destination) serial=6601 path=/; interface=org.freedesktop.fwupd; member=Changed
```
</details>
[image]

elioqoshi, 2022-10-04:
Modals look good to me. Maybe signal the level of significance with color more (red in this case as it's critical). Also title and description should both be left aligned:

[image]

Jongmassey, 2025-02-07:
`User has configured their system in a broken way`  this is deeply unhelpful. What have I configured in a broken way? This is out of the box Ubuntu 24.04.1 LTS. Doesn't feel very "LTS"