canonical/firmware-updater #69: Update Available pill hides the device name

Feichtmeier, 2022-09-22:
[image]

An idea would be to exchange the subtile with the pill if upgradeAvailable :shrug: 

[image]

jpnurmi, 2022-09-22:
would it make sense to replace the device icon with a green "update" icon?

Feichtmeier, 2022-09-22:
That would be cool only needs a change in yaruwidgets so let the normal master detail page make use of the yarupageitem itemWidget property (which is currently only being used in the compact layout)
But I think to remember that elio almost always prefers text over icons :P for usability accessibility reasons =)

elioqoshi, 2022-10-04:
> would it make sense to replace the device icon with a green "update" icon?

I'd not say so. Iconography should not replace label

> An idea would be to exchange the subtitle with the pill if upgradeAvailable 🤷

Can't the pill be added underneath the subtitle and expand the row height? That would be cleaner.

Also I think if the label is simply "Update" that would suffice.