canonical/firmware-updater #70: BackButton could have a better form

Feichtmeier, 2022-09-22:
The back button for nested pages is a squeezed pill (prbly caused by the reduced appbar height in yaru.dart...)
[image]

A solution would be to put it inside an InkWell instead of an IconButton
[image]

jpnurmi, 2022-09-22:
hmm, perhaps we could also set `elevation: 0` to remove that extra line until scrolled under?

Feichtmeier, 2022-09-22:
This un-even circle (alias... ellipse :D ) looks a bit lost in this corner of two lines nonetheless IMO. The appbar border would not change this I think

PR #75 (2022-09-23): ReleasePage: tweak app bar & back button looks
- remove the separator until scrolled under to give it less boxy looks
- ensure that the back button stays round

[image]

[image]

Close: #70
commit 96424d2d (2022-09-23): Fix back button shape (#75)

Close: #70
closed 2022-09-23 by jpnurmi