canonical/firmware-updater #80: Progress visualization during installation

jpnurmi, 2022-09-28:
A spinner or a linear progress bar? Where should we show it?

elioqoshi, 2022-10-04:
Think we went over this in the call but it would make sense to have a linear progress bar (as long as we can indicate the absolute progress amount) and also replace the button with the progress as a label (the button would be technically disabled as well)

d-loose, 2022-10-13:
I've re-added the old app bar with its linear progress indicator in #102. As mentioned in the comments there, we should have a way of telling the user what's currently going on. Here's my current draft, showing the status in the app bar:

[Screencast from 13-10-22 11:42:37.webm](https://user-images.githubusercontent.com/113362648/195563702-9d3f629b-08e0-43a1-812b-61dd3aa33efa.webm)

While I think this looks nice with a wide window, it doesn't work if there's not enough horizontal space. Does anyone have suggestions for better placement?

jpnurmi, 2022-10-13:
How would it look with a font similar to the subtitle?

jpnurmi, 2022-10-13:
Brainstorming ideas:
- move the progress bar and status text in a "banner" below the app bar
- a status bar at the bottom

d-loose, 2022-10-13:
A smaller font is definitely necessary and helps:

[image]

> Brainstorming ideas:
> 
>     * move the progress bar and status text in a "banner" below the app bar
> 
>     * a status bar at the bottom

I'd prefer the first option, since I like keeping the status updates and progress bar close to the disabled buttons - I'm not a UI/UX design expert (yet!), but this is where I'd expect those.
@elioqoshi do you have any feedback? :)