canonical/firmware-updater #81: Device action buttons

jpnurmi, 2022-09-28:
Should we move the buttons at the bottom?

- Vendor name, device name, or device description could be relatively long
- There can be up to three buttons
- The device page buttons are at the top
- The release page buttons are at the bottom

[image]
[image]

Feichtmeier, 2022-09-28:
I would say yes if you keep the header to be this very large 
Alternatively you could shrink the headers to ba a normal app bar and put all buttons beneath the appbar at the top

elioqoshi, 2022-10-04:
@Feichtmeier's suggestioin sounds interesting, though I have to see it better to have a better idea of it.

I think it might be better if the release page buttons are at the top as well but will have to test it to ensure nothing breaks horribly

jpnurmi, 2022-10-11:
> Keep them right-aligned but below the app bar?
:+1: