canonical/firmware-updater #112: Multiple reboot confirmation dialogs

jpnurmi, 2022-10-19:
I run a system firmware upgrade that requires a system restart. We'll need to move the reboot dialog out of `build()` because they would pile up if there are multiple change notifications (due to #105?) while in reboot mode.

[image]
[image]
[image]

A possibly related exception was also thrown under the hood:
```
Launching lib/main.dart on Linux in debug mode...
Building Linux application...                                           
Syncing files to device Linux...                                    64ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on Linux is available at: http://127.0.0.1:35683/204CZobDDS4=/
The Flutter DevTools debugger and profiler on Linux is available at: http://127.0.0.1:9100?uri=http://127.0.0.1:35683/204CZobDDS4=/
flutter: DEBUG device_store: 9 devices
flutter: DEBUG fwupd_notifier: on battery: false
flutter: DEBUG fwupd_service: install FwupdRelease(appstreamId: com.dell.uefi398c4c37.firmware, checksum: 78dca60eee38bdf71362cb83eeabc481f90167f9,91a692607c6f21ab123c05afd6aa5a3b2db3303304282f2e2354f198f9e44ea0, created: 2022-09-12 00:00:00.000Z, description: '<p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address security vulnerabilities.</li></ul><p>Do not turn off your computer or remove the AC adapter while the update is in progress.</p>', filename: Dell_XPS_15_9520___Precision_5570_System_BIOS_ver.1.7.0.cab, homepage: http://support.dell.com/, license: LicenseRef-proprietary, locations: [https://fwupd.org/downloads/91a692607c6f21ab123c05afd6aa5a3b2db3303304282f2e2354f198f9e44ea0-Dell_XPS_15_9520___Precision_5570_System_BIOS_ver.1.7.0.cab], name: 'XPS 15 9520, Precision 5570', protocol: org.uefi.capsule, remoteId: lvfs, size: 23531955, flags: {FwupdReleaseFlag.isUpgrade}, urgency: FwupdReleaseUrgency.critical, uri: https://fwupd.org/downloads/91a692607c6f21ab123c05afd6aa5a3b2db3303304282f2e2354f198f9e44ea0-Dell_XPS_15_9520___Precision_5570_System_BIOS_ver.1.7.0.cab, vendor: 'Dell', version: '1.7.0')
flutter: DEBUG fwupd_service: on FwupdDevice(checksum: 6a19309e3006368629360bedbf22979b487954ff769ffd43338e18e3cfbd6cf5, created: 2022-10-19 06:00:36.000Z, deviceId: a45df35ac0e948ee180fe216a5f703f32dda163f, flags: {FwupdDeviceFlag.internal, FwupdDeviceFlag.updatable, FwupdDeviceFlag.requireAc, FwupdDeviceFlag.supported, FwupdDeviceFlag.registered, FwupdDeviceFlag.needsReboot, FwupdDeviceFlag.canVerify, FwupdDeviceFlag.usableDuringUpdate}, guid: [398c4c37-8057-4fe2-acaa-c567937c61b8, 230c8b18-8d9b-53ec-838b-6cfc0383493a], icon: [computer], modified: null, name: 'System Firmware', parentDeviceId: null, plugin: uefi_capsule, protocol: org.uefi.capsule, summary: UEFI ESRT device, updateState: FwupdUpdateState.success, vendor: Dell, vendorId: DMI:Dell Inc., version: 1.6.0, versionBootloader: null, versionFormat: FwupdVersionFormat.dellBios, versionLowest: 1.6.0)
flutter: DEBUG fwupd_service: download https://fwupd.org/downloads/91a692607c6f21ab123c05afd6aa5a3b2db3303304282f2e2354f198f9e44ea0-Dell_XPS_15_9520___Precision_5570_System_BIOS_ver.1.7.0.cab to /tmp/91a692607c6f21ab123c05afd6aa5a3b2db3303304282f2e2354f198f9e44ea0-Dell_XPS_15_9520___Precision_5570_System_BIOS_ver.1.7.0.cab
flutter: DEBUG fwupd_notifier: 0%
flutter: DEBUG fwupd_notifier: 1%
flutter: DEBUG fwupd_notifier: 2%
flutter: DEBUG fwupd_notifier: 3%
flutter: DEBUG fwupd_notifier: 4%
flutter: DEBUG fwupd_notifier: 5%
flutter: DEBUG fwupd_notifier: 6%
flutter: DEBUG fwupd_notifier: 7%
flutter: DEBUG fwupd_notifier: 8%
flutter: DEBUG fwupd_notifier: 9%
flutter: DEBUG fwupd_notifier: 10%
flutter: DEBUG fwupd_notifier: 11%
flutter: DEBUG fwupd_notifier: 12%
flutter: DEBUG fwupd_notifier: 13%
flutter: DEBUG fwupd_notifier: 14%
flutter: DEBUG fwupd_notifier: 15%
flutter: DEBUG fwupd_notifier: 16%
flutter: DEBUG fwupd_notifier: 17%
flutter: DEBUG fwupd_notifier: 18%
flutter: DEBUG fwupd_notifier: 19%
flutter: DEBUG fwupd_notifier: 20%
flutter: DEBUG fwupd_notifier: 21%
flutter: DEBUG fwupd_notifier: 22%
flutter: DEBUG fwupd_notifier: 23%
flutter: DEBUG fwupd_notifier: 24%
flutter: DEBUG fwupd_notifier: 25%
flutter: DEBUG fwupd_notifier: 26%
flutter: DEBUG fwupd_notifier: 27%
flutter: DEBUG fwupd_notifier: 28%
flutter: DEBUG fwupd_notifier: 29%
flutter: DEBUG fwupd_notifier: 30%
flutter: DEBUG fwupd_notifier: 31%
flutter: DEBUG fwupd_notifier: 32%
flutter: DEBUG fwupd_notifier: 33%
flutter: DEBUG fwupd_notifier: 34%
flutter: DEBUG fwupd_notifier: 35%
flutter: DEBUG fwupd_notifier: 36%
flutter: DEBUG fwupd_notifier: 37%
flutter: DEBUG fwupd_notifier: 38%
flutter: DEBUG fwupd_notifier: 39%
flutter: DEBUG fwupd_notifier: 40%
[... 84 lines cut ...]
flutter: DEBUG fwupd_notifier: FwupdStatus.idle
flutter: DEBUG fwupd_notifier: 100%
flutter: DEBUG fwupd_notifier: BatteryLevel changed
flutter: DEBUG fwupd_notifier: BatteryThreshold changed
flutter: DEBUG fwupd_notifier: OnlyTrusted changed
flutter: DEBUG device_store: 9 devices
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Looking up a deactivated widget's ancestor is unsafe.
At this point the state of the widget's element tree is no longer stable.
To safely refer to a widget's ancestor in its dispose() method, save a reference to the ancestor by calling dependOnInheritedWidgetOfExactType() in the widget's didChangeDependencies() method.
#0      Element._debugCheckStateIsActiveForAncestorLookup.<anonymous closure> (package:flutter/src/widgets/framework.dart:4241:9)
#1      Element._debugCheckStateIsActiveForAncestorLookup (package:flutter/src/widgets/framework.dart:4255:6)
#2      Element.dependOnInheritedWidgetOfExactType (package:flutter/src/widgets/framework.dart:4275:12)
#3      Localizations.of (package:flutter/src/widgets/localizations.dart:472:48)
#4      AppLocalizations.of (package:flutter_gen/gen_l10n/app_localizations.dart:67:26)
#5      showMessageDialog (package:firmware_updater/src/widgets/confirmation_dialog.dart:18:33)
#6      showConfirmationDialog (package:firmware_updater/src/widgets/confirmation_dialog.dart:88:5)
#7      DevicePage.build.<anonymous closure> (package:firmware_updater/device_page.dart:69:16)
#8      SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1175:15)
#9      SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1113:9)
#10     SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:1015:5)
#11     _invoke (dart:ui/hooks.dart:148:13)
#12     PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:318:5)
#13     _drawFrame (dart:ui/hooks.dart:115:31)

```

d-loose, 2022-10-19:
Thanks for spotting this - it would be great to include dialog tests in the integration tests, once this is resolved.

> I run a system firmware upgrade on 22.10. The upgrade requires a system restart, which resulted in three confirmation dialogs stacked on top of each other. We'll need to move the reboot dialog out of `build()` because they would pile up if there are multiple change notifications (due to #105?) while in reboot mode.

I've done some testing and it seems the multiple build calls are due to
```dart
await notifier.refresh()
```
Would it be a terrible solution to keep track of whether a dialog is currently shown in `DeviceModel` and make sure to only show one?

> A possibly related exception was also thrown under the hood

This one is caused by
```dart
await store.refresh()
```
which causes the entire app to re-draw, destroying the build context of the dialog. This is a terrible workaround anyways.. :/

jpnurmi, 2022-10-19:
What if we would pass an `onError` callback from `initState` that would allow us to react once when it occurs? In general, we need to be prepared that `build()` may get called for unrelated reasons. Window metrics, locale/language, system theme, or anything alike could trigger rebuilds that are not related to fwupd or model state changes. :)

jpnurmi, 2022-10-19:
I volunteer to complicate things! :raised_back_of_hand: Now that I think of it, we should be able to present errors even if the user navigates to another page meanwhile. Are we able to lift error handling up in the tree?

d-loose, 2022-10-19:
Ah good point - currently the user doesn't see any errors if they navigate to another page during an update, which isn't great. I added the error handling to the individual `DeviceModel`s thinking we would block navigating to another device as well during the installation.
But I like the idea of leaving the rest of the UI usable. We'd have to handle errors at the level of the `FirmwareApp` then, probably in what is currently called `DeviceStore`.

d-loose, 2022-10-27:
@jpnurmi can you confirm that this has been resolved by #115?

jpnurmi, 2022-10-27:
Looks good! You can mention "Close: #NNN" in the PR that fixes an issue to automatically close it out of the way. :)

closed 2022-10-27 by jpnurmi