canonical/firmware-updater #154: Should we use YaruWindowTitleBar?

jpnurmi, 2023-02-20:
I'm not sure if replacing `GtkHeaderBar` by `YaruWindowTitleBar` alone is worth it but perhaps we could make better use of the window title bar if we used the Flutter-based one?

| GtkHeaderBar | YaruTitleBar |
|---|---|
| [image] | [image] |

jpnurmi, 2023-02-20:
For example:
[image]

jpnurmi, 2023-02-20:
P.S. I took a shortcut and ignored the battery warning banner because with this structure we cannot easily have it on both sides of master-detail and between title and content, at the same time. Perhaps we could move it to the bottom, though?

[image]

d-loose, 2023-02-20:
Yes, I think it looks much better like this!

d-loose, 2023-02-22:
One issue I found while playing around with this: how do we navigate back in the portrait layout? Intuitively I'd expect a back button as a leading widget in the title bar, but this requires access to the MD page's navigator.

[Screencast from 2023-02-22 11-58-57.webm](https://user-images.githubusercontent.com/113362648/220601182-2d46ac1b-0ac0-4d6d-b8b1-d0bfaf4934e1.webm)

Feichtmeier, 2023-02-22:
Oh dear 😸 

I think we are facing a similar issue like in software. I could propopse to use the same layout as in Software which is aestecially pleasing but makes the code complicated again. At least it provides a back button

PR #156 (2023-02-27): use YaruWindowTitleBar
Does centering the title on the right make it better or worse? I think the space between the icon and the name looks a bit weird.

[Screencast from 2023-02-22 12-28-15.webm](https://user-images.githubusercontent.com/113362648/220607579-e0211760-a83a-4d9d-8993-27692f5adf23.webm)

Fix #154
closed 2023-02-27 by d-loose