canonical/firmware-updater #164: Denial when calling reboot in the snap

kenvandine, 2023-03-29:
Reboot is failing for me due to this denial:

2023-03-29T09:36:21.844452-04:00 lifebook dbus-daemon[1835]: apparmor="DENIED" operation="dbus_method_call"  bus="session" path="/org/gnome/SessionManager" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name="org.gnome.SessionManager" pid=4948 label="snap.firmware-updater.firmware-updater-app" peer_pid=1991 peer_label="unconfined"
[image]

d-loose, 2023-03-29:
It seems like this isn't covered by the `desktop` interface. The only interface that partially allows access to `org.gnome.SessionManager` is [`screen-inhibit-control`](https://github.com/snapcore/snapd/blob/6b29eb8fae209982a1d162e4f9fba19a0163ffb7/interfaces/builtin/screen_inhibit_control.go#L36-L42), which gives access to `Inhibit` and `Uninhibit`.

We'd still be able to reboot using `org.freedesktop.login1.Manager` though (but this way the user won't see a confirmation dialog). Access to that is already provided by [`fwupd`](https://github.com/snapcore/snapd/blob/6b29eb8fae209982a1d162e4f9fba19a0163ffb7/interfaces/builtin/fwupd.go#L209-L214)

shytzedaka, 2023-09-26:
i have a similar issue:
[image]

it appears that i am unable to apply a firmware update.
although i am guessing such a program like firmware updater would require elevated permissions?
i tried opening it with sudo in the hopes that might bypass any permission issues however when trying to launch it with sudo i was greeted with: 
`mkdir: cannot create directory ‘/run/user/0’: Permission denied
unable to connect to D-Bus: Failed to execute child process “dbus-launch” (No such file or directory)`

closed 2023-10-10 by d-loose