canonical/firmware-updater #177: UI improvements

d-loose, 2023-06-06:
The current UI needs some improvements.

Screenshots:
| | Light | Dark |
| - | - | - |
| Device page | [image] | [image] |
| Release Page | [image] | [image] |
| Modals | [image] | [image] |

elioqoshi, 2023-06-08:
Did some updates and presented this in a little nicer way: 

[image]

hellsworth, 2023-06-08:
Maybe also add a little italic blurb below the "Refresh Checksums" saying what that means for the non techie users.

d-loose, 2023-06-29:
@elioqoshi something I forgot to ask:
> Introduced a new update available label when there is a newer version available

[image]

There's no such bright version of the accent color in the yaru palette - should it just be this specific color independent of the rest of the theme?

elioqoshi, 2023-06-29:
@d-loose you mean there is no background color green?

Ideally there would be a static Tin color in Yaru, which I proposed in the color file a few weeks ago:

[image]

If not, you could simulate the tint color by giving the primary color opacity, e;g 10%. It will not be exactly the same but close. I'd prefer if Yaru has separate color swatches for these situations though as dealing with opacity is not very sustainable.

d-loose, 2023-06-29:
Thanks, I'll see what that looks like.
Looking forward to the Yaru colorscheme overhaul! :)

d-loose, 2023-06-29:
With 10% opacity:

|Light|Dark|
|-|-|
|[image]|[image]|

Contrast is quite low in the dark version