canonical/firmware-updater #180: Missing dock and alt-tab icon

tim-hm, 2023-06-06:
## Background

When running firmware-updater on lunar the dock and super+tab icons are missing (I can't capture a screenshot of the super+tab).

- I am running the firmware-updater from latest/edge `0+git.60bef69 2023-06-03`
- I am on lunar lobster

## Missing Icon In Dock

[image]

jbicha, 2023-09-15:
@kenvandine Just a reminder that this is still an issue with Ubuntu 23.10

tim-hm, 2023-09-18:
@d-loose can you add this to your backlog?

d-loose, 2023-09-18:
Let's wait for @kenvandine's reply - I talked with him about this before and he said he'd look into it, since it's a somewhat tricky xdg icon naming issue

kenvandine, 2023-09-18:
@sergio-costas can you please take a look at this?  It's happening because there's a shell script that launches it via gapplication and I wasn't able to figure out why it doesn't match an icon.  I guess it has something to do with the Exec not matching the process that's running?

sergio-costas, 2023-09-18:
I'll check it.

sergio-costas, 2023-09-18:
Finally, found how to fix it. Sending a patch.

closed 2023-09-18 by tim-hm