canonical/firmware-updater #207: Firmware-updater can't reboot the system

sergio-costas, 2023-10-11:
Today there was an update for my system, and firmware-updater asked to reboot. I pressed "Reboot" and I received this error:

[image]

It seems that firmware-updater snap lacks the shutdown interface: https://snapcraft.io/docs/shutdown-interface

sergio-costas, 2023-10-11:
Also, it seems to use a different DBus interface: while the "shutdown" interface uses the "org.freedesktop.login1.Manager" interface in the system bus, firmware-updater wants to use the "org.gnome-SessionManager" interface in the session bus.

sergio-costas, 2023-10-11:
So there are two possibilities:

* either change the shutdown method in firmware-updater to use the "org.freedesktop.login1.Manager" interface
* or modify snapd to also allow to access the org.gnome.SessionManager interface when the "shutdown" interface is connected

In any case, it will be required to add the "shutdown" plug to firmware-updater, and also request to the snap store to grant the automatic connection of this interface.

d-loose, 2023-10-11:
Ah, forgot to close this issue as well - fixed in #206.

sergio-costas, 2023-10-11:
No, you didn't forget; I filled it just ten minutes ago :-D

Thanks.

d-loose, 2023-10-11:
Ah, sorry too many issues, I got confused :D

d-loose, 2023-10-11:
> In any case, it will be required to add the "shutdown" plug to firmware-updater, and also request to the snap store to grant the automatic connection of this interface.

This is an important point though - I just tested the latest snap in the candidate channel, and the `shutdown` plug doesn't automatically connect. Could you request this?

sergio-costas, 2023-10-11:
Mmm... but the snap doesn't have the "shutdown" plug... And if the "fwupd" plug already includes access to that DBus interface, then it's not needed.

d-loose, 2023-10-11:
I've added the `shutdown` plug in #206. As far as I understand the `fwupd` only works if the corresponding slot is provided by a snap, which isn't the case yet (see details and links in #206).

sergio-costas, 2023-10-11:
Oh, my fault.

sergio-costas, 2023-10-11:
Request sent: https://forum.snapcraft.io/t/request-for-firmware-update-shutdown-plug-autoconnection/37270

Feichtmeier, 2023-10-24:
This worked for me when I switched to the latest edge channel with the reboot
👍

d-loose, 2023-10-24:
Thanks for confirming!
Resolved by #206.

closed 2023-10-24 by d-loose