canonical/firmware-updater #209: firmware-updater can't run with fwupd snap

superm1, 2023-10-11:
I just recently installed Ubuntu 23.10 beta, and wanted to experiment with the `firmware-updater` snap.    It seems to work fine with the integrated `fwupd` debian package.

However when I upgrade to the confined snap version of `fwupd`, the `firmware-updater` app hangs here:

[image]

To install the fwupd snap, these were my steps:
```
sudo apt purge fwupd
sudo snap install fwupd
```

I made sure the fwupd snap worked with:
```
fwupd.fwupdmgr get-devices
```

Here are the versions of the various apps:
```
$ snap info fwupd firmware-updater
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   unset
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
commands:
  - fwupd.dbxtool
  - fwupd.fwupdmgr
  - fwupd.fwupdtool
services:
  fwupd: dbus, enabled, active
snap-id:      HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
tracking:     latest/stable
refresh-date: today at 12:25 CDT
channels:
  latest/stable:    1.9.6               2023-10-09 (5236) 24MB -
  latest/candidate: 1.9.6               2023-10-06 (5236) 24MB -
  latest/beta:      ↑
  latest/edge:      1.9.6-16-gcee1efd06 2023-10-11 (5246) 24MB -
installed:          1.9.6                          (5236) 24MB -
---
name:      firmware-updater
summary:   Firmware Updater
publisher: Canonical✓
store-url: https://snapcraft.io/firmware-updater
license:   GPL-3.0+
description: |
  Update Firmware
commands:
  - firmware-updater
services:
  firmware-updater.firmware-notifier:    simple, enabled, inactive
  firmware-updater.firmware-updater-app: dbus, enabled, inactive
snap-id:      EI0D1KHjP8XiwMZKqSjuh6W8zvcowUVP
tracking:     latest/stable/ubuntu-23.10
refresh-date: 22 days ago, at 18:03 CDT
channels:
  latest/stable:    0+git.e8771be 2023-09-19 (109) 11MB -
  latest/candidate: 0+git.773f947 2023-10-10 (111) 11MB -
  latest/beta:      ↑
  latest/edge:      0+git.019bf66 2023-07-17 (105) 11MB -
installed:          0+git.e8771be            (109) 11MB -

$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.6
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.6
runtime   org.kernel                    6.5.0-9-generic
```

d-loose, 2023-10-12:
Thanks! I just tested this - it seems like there are apparmor denials in that case:

```
Oct 12 09:40:22 ubuntu-pc-i440fx-mantic kernel: audit: type=1107 audit(1697103622.114:189): pid=465 uid=101 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/" interface="org.freedesktop.fwupd" member="GetDevices" mask="send" name="org.freedesktop.fwupd" pid=8831 label="snap.firmware-updater.firmware-updater-app" peer_pid=4351 peer_label="snap.fwupd.fwupd"
                                                exe="/usr/bin/dbus-daemon" sauid=101 hostname=? addr=? terminal=?'
Oct 12 09:40:22 ubuntu-pc-i440fx-mantic kernel: audit: type=1107 audit(1697103622.114:190): pid=465 uid=101 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/" interface="org.freedesktop.fwupd" member="GetDevices" name=":1.147" mask="receive" pid=4351 label="snap.fwupd.fwupd" peer_pid=8831 peer_label="snap.firmware-updater.firmware-updater-app"
                                                exe="/usr/bin/dbus-daemon" sauid=101 hostname=? addr=? terminal=?'
```

superm1, 2023-10-12:
@valentindavid can you take a look at this?  Is it something we need different in the fwupd snap to allow this path to work?

rebastion, 2023-10-16:
while this is being fixed possibly, is there a quick workaround to get the firmware-updater app running?

superm1, 2023-12-21:
> while this is being fixed possibly, is there a quick workaround to get the firmware-updater app running?
(edit: doesn't seem possible to remove the fwupd snap)

You can shift back from snap to deb until they can figure out a fix for this.
```shell
sudo snap remove fwupd
sudo apt install fwupd
```

FWIW; this also affects the Ubuntu software store snap.  So whatever the solution is needs to be applied to both:

[image]

freebeego, 2023-12-28:
+