canonical/firmware-updater #215: "closable" AC power warning is in a confusing location

superm1, 2023-10-16:
When I run `Firmware Updater` without an AC adapter plugged in I get a warning in the top of the window that some stuff can't be updated without AC.

I think the warning itself is good to have, but I think having an X for the warning in the same place that you would usually "close the window" is confusing.  I think it would be better to put this somewhere else prominent; like the bottom of the window.

[image]

Glen-Campbell-Bose, 2024-01-16:
I agree with this and have put together some code to make the warning message into an AlertDialog. Is that acceptable? 
There's another case where if the BIOS boot mode is "Legacy", you can't update. I've created an AlertDialog for this as well. Is that a separate issue, or can I include that with this issue?

[image]

superm1, 2024-01-16:
That looks great to me!

superm1, 2024-01-16:
IMO For BIOS mode of legacy I think it should have an option to waive in the future/ignore it.  That can be expected for some people and there isn't much they can do.

Glen-Campbell-Bose, 2024-01-16:
That's a good idea. A little check-box on the warning "don't show this in the future"?
Although I tripped over this and it motivated me to reinstall everything as UEFI-bootable.
It was a pain in the neck, but I was glad I did it.

superm1, 2024-01-16:
> That's a good idea. A little check-box on the warning "don't show this in the future"?
> Although I tripped over this and it motivated me to reinstall everything as UEFI-bootable.
> It was a pain in the neck, but I was glad I did it.

Yeah something like that.

There's other things that could be useful to prompt and save a setting like this in the future too so I suggest building it into a reusable snippet or function.

The other thing that comes to mind is opting in to reports after an update.

Glen-Campbell-Bose, 2024-01-16:
OK.