canonical/firmware-updater #236: Conflict with TPM-backed Full Disk Encryption

jbicha, 2023-11-29:
See this report:
https://discourse.ubuntu.com/t/ubuntu-23-10-asking-for-tpm-recovery-key-on-every-boot-after-firmware-update/40656

superm1, 2023-12-21:
Related is this issue with deb fwupd: https://github.com/canonical/ubuntu-desktop-installer/issues/2371 and this one with snap fwupd: https://github.com/fwupd/fwupd/issues/6264

zacdirect, 2025-10-04:
+1 on this I suppose.  I get this same issue on a Dell Max 16.

[image]

Having gone some rounds on this for other hardware, it may just be that there are now 'update' lanes for this kind of thing, as opposed to direct overwrites? 

https://askubuntu.com/questions/1534463/unable-to-install-sof-firmware-on-ubuntu-24-04/1556831#1556831