canonical/firmware-updater #255: Issue submitted via apport

schopin-pro, 2024-01-17:
**ProblemType**
Bug

**Date**
Wed Jan 17 15:58:42 2024

**CurrentDesktop**
GNOME

**ProcEnviron**
LANG=fr_FR.UTF-8
PATH=(custom, no user)
SHELL=/usr/bin/zsh
TERM=xterm-256color
XDG_RUNTIME_DIR=<set>

**DistroRelease**
Ubuntu 23.10

**Uname**
Linux 6.5.0-14-generic x86_64

**Architecture**
amd64

**Snap**
firmware-updater 0+git.e8771be (latest/stable)

**SnapChanges**
aucun changement trouvé

**SnapConnections**
Interface               Connecteur                       Prise                           Notes
content[gnome-42-2204]  firmware-updater:gnome-42-2204   gnome-42-2204:gnome-42-2204     -
content[gtk-3-themes]   firmware-updater:gtk-3-themes    gtk-common-themes:gtk-3-themes  -
content[icon-themes]    firmware-updater:icon-themes     gtk-common-themes:icon-themes   -
content[sound-themes]   firmware-updater:sound-themes    gtk-common-themes:sound-themes  -
dbus                    -                                firmware-updater:dbus-slot      -
desktop                 firmware-updater:desktop         :desktop                        -
desktop-legacy          firmware-updater:desktop-legacy  :desktop-legacy                 -
fwupd                   firmware-updater:fwupd           :fwupd                          -
gsettings               firmware-updater:gsettings       :gsettings                      -
opengl                  firmware-updater:opengl          :opengl                         -
upower-observe          firmware-updater:upower-observe  :upower-observe                 -
wayland                 firmware-updater:wayland         :wayland                        -
x11                     firmware-updater:x11             :x11                            -

**SnapInfo.firmware-updater**
name:      firmware-updater
summary:   Firmware Updater
publisher: Canonical**
store-url: https://snapcraft.io/firmware-updater
contact:   https://github.com/canonical/firmware-updater/
license:   GPL-3.0+
description: |
  Update Firmware
commands:
  - firmware-updater
services:
  firmware-updater.firmware-notifier:    simple, enabled, inactive
  firmware-updater.firmware-updater-app: dbus, enabled, inactive
snap-id:      EI0D1KHjP8XiwMZKqSjuh6W8zvcowUVP
tracking:     latest/stable/ubuntu-23.10
refresh-date: 2023-11-03T10:26:16+01:00
channels:
  latest/stable:    0+git.e8771be 2023-09-19T12:58:41Z (109) 11MB -
  latest/candidate: 0+git.e1cd612 2023-12-13T10:27:44Z (120) 11MB -
  latest/beta:      ^                                             
  latest/edge:      0+git.019bf66 2023-07-17T17:43:22Z (105) 11MB -
installed:          0+git.e8771be                      (109) 11MB -
[... 131 lines cut ...]
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req vnmi avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg tme avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear ibt flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs eibrs_pbrsb gds
bogomips	: 5990.40
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

**ApportVersion**
2.27.0-0ubuntu5

**CasperMD5CheckResult**
pass

schopin-pro, 2024-01-17:
Well, the failed update might be something else entirely, as I'm getting the following errors in journalctl, even when selecting the version manually:

```
janv. 17 16:05:29 pug kernel: audit: type=1326 audit(1705503929.102:2100): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.firmware-updater.firmware-updater-app pid=82593 comm="DartWorker" exe="/snap/firmware-updater/109/bin/firmware-updater" sig=0 arch=c000003e syscall=41 compat=0 ip=0x7f399ab58d2b code=0x50000
janv. 17 16:05:29 pug kernel: audit: type=1326 audit(1705503929.118:2101): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.firmware-updater.firmware-updater-app pid=82593 comm="DartWorker" exe="/snap/firmware-updater/109/bin/firmware-updater" sig=0 arch=c000003e syscall=41 compat=0 ip=0x7f399ab58d2b code=0x50000
janv. 17 16:05:29 pug kernel: audit: type=1326 audit(1705503929.862:2102): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.firmware-updater.firmware-updater-app pid=82593 comm="DartWorker" exe="/snap/firmware-updater/109/bin/firmware-updater" sig=0 arch=c000003e syscall=41 compat=0 ip=0x7f399ab58d2b code=0x50000
janv. 17 16:05:29 pug kernel: audit: type=1326 audit(1705503929.874:2103): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.firmware-updater.firmware-updater-app pid=82593 comm="DartWorker" exe="/snap/firmware-updater/109/bin/firmware-updater" sig=0 arch=c000003e syscall=41 compat=0 ip=0x7f399ab58d2b code=0x50000
janv. 17 16:05:34 pug kernel: audit: type=1107 audit(1705503934.018:2104): pid=2108 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.14" mask="receive" pid=83333 label="snap.firefox.firefox" peer_pid=2167 peer_label="unconfined"
                               exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
janv. 17 16:05:34 pug kernel: audit: type=1107 audit(1705503934.034:2105): pid=2108 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.14" mask="receive" pid=83333 label="snap.firefox.firefox" peer_pid=2167 peer_label="unconfined"
                               exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
```

berkus, 2026-11-09:
Hi, I could not reproduce this on the current version. If you have any way to repro this, please reopen and I'll give it another look. Thank you!

closed 2026-11-09 by berkus