canonical/firmware-updater #292: App name in desktop file not translatable

AsciiWolf, 2024-05-23:
It looks like that the "Firmware Updater" app name is always in English in GNOME Shell Overview although it is fully translated.

[image]

AsciiWolf, 2024-05-25:
Here is an example of how it would look for the Czech language:

```
diff --git a/packages/firmware_updater/assets/firmware-updater.desktop b/packages/firmware_updater/assets/firmware-updater.desktop
index 7a8d875..16f1c1c 100644
--- a/packages/firmware_updater/assets/firmware-updater.desktop
+++ b/packages/firmware_updater/assets/firmware-updater.desktop
@@ -2,7 +2,9 @@
 Type=Application
 Version=1.0
 Name=Firmware Updater
+Name[cs]=Aktualizátor firmwaru
 Comment=Manage firmware on your device
+Comment[cs]=Spravujte firmware na vašem zařízení
 Keywords=firmware;
 #use sh because pkexec is broken under xfce/lxce http://pad.lv/1193526
 Exec=firmware-updater
```

Feel free to apply these changes or if you wish, I can make a Pull request. ;-)

AsciiWolf, 2024-06-18:
/cc @d-loose

AsciiWolf, 2024-12-08:
/cc @d-loose

Still the same problem.

FAlexei, 2025-01-12:
@AsciiWolf Could you please create a Pull Request for the Russian language?

I tried to create it myself (#305), but I was asked to register.
Also, that PR had a failed check. Could it be that adding this type of translations is not possible?

Name[ru]=Обновление прошивок
Comment[ru]=Управляйте прошивками своих устройств

AsciiWolf, 2025-01-12:
@FAlexei That's not up to me. I can create the PR, but it probably won't get merged.

The problem needs to be discussed first and a solution approved.

AsciiWolf, 2025-04-17:
Still the same problem in Ubuntu 25.04.

AsciiWolf, 2026-02-06:
Still the same problem with latest firmware-updater Snap.