canonical/firmware-updater #297: User agent doesn't report to LVFS correctly

superm1, 2024-07-04:
The LVFS web service has requirements for clients to report their user agents in a structured fashion and to include specific information.  Those requirements are summarized [here](https://lvfs.readthedocs.io/en/latest/download.html).  The reasons for those requirements are also at the same link.

Firmware-updater is an outlier that is not obeying the requirements and is only reporting the user agent:

```
firmware-updater
```

Normally a user agent will report:
* client name, e.g. gnome-software/46.2
* system description, e.g. Linux x86_64
* user language, e.g. en-US
* operating system, e.g. Fedora Linux 40
* the fwupd daemon runtime version, e.g. fwupd/1.9.21

If this isn't solved; then LVFS may eventually block the firmware-updater user agent string.

superm1, 2024-10-08:
@d-loose can you please fix this?  Like I said LVFS is going to cut off non structured user agents and it will totally break this client.

d-loose, 2024-10-08:
Thanks for the ping - will take a look this week!

PR #313 (2024-10-09): feat: make user agent string comply with LVFS requirements
Fix #297
UDENG-4906
commit a80e666d (2025-01-30): feat: make user agent string comply with LVFS requirements

Fix #297
UDENG-4906
closed 2024-10-09 by d-loose