canonical/firmware-updater #314: firmware-notifier sends too many desktop notifications, and there is no way to silence it.

KlipperKyle, 2024-10-09:
When a firmware update is available, this application (specifically firmware-notifier) sends a desktop notification once every few hours until you update the system firmware.  The application should notify only once, if that.  Anything beyond that is nagging behavior.

(Personally, I would prefer to turn off notifications entirely, because like most experienced users I dedicate a specific time to perform these kinds of upgrades.)

bmaupin, 2024-12-13:
This was annoying me too. I tried disabling the `fwupd` service, but it appears that firmware-updater will simply start it again at the next boot. You might be able to mask the `fwupd` service so that it can't be started, but then firmware-updater won't work at all.

What I wanted instead was to be able to update firmware on demand, but not have an annoying popup multiple times a day.

Next I went into the Gnome settings and I saw there's an option to disable notifications for individual applications, but unfortunately disabling the notifications for firmware-updater doesn't have any effect. Maybe because it's a snap it doesn't integrate well with Gnome.

I tried one other thing and it seems to be working so far; firmware-updater seems to run as a snap service, and it seems like it can be stopped:

```
$ sudo snap stop --disable firmware-updater.firmware-notifier

$ snap services
Service                                              Startup   Current   Notes
firmware-updater.firmware-notifier                   disabled  inactive  user,timer-activated
firmware-updater.firmware-updater-app                enabled   inactive  user,dbus-activated
```

KlipperKyle, 2024-12-14:
Since this bug has yet to get the attention and triage of a Canonical engineer, here is a simpler workaround that worked for me:

```
sudo snap remove firmware-updater
```

And you can update manually with fwupdmgr, as God intended.

```
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr upgrade
```

Side note: when I applied the firmware upgrade in question (the one of which firmware-updater's nag prompted me to open this bug report), it almost bricked my laptop and required a power drain to get the poor thing to POST again.

Dear maintainer, it is a *very* poor user experience to nag the user several times per day until he applies an update which has the potential to brick his machine.  Please fix your software.