canonical/firmware-updater #362: Current Version is not correctly displayed if a patch version starts with 0

Lidoca, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues and found none matching what I'm reporting.

### Bug Description

The example is about the following firmware: <https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2LHT.firmware>

[image]

### Steps to reproduce the behavior

1. Prepare any device for which a firmware update or downgrade is available, where the patch version starts with 0.
   - Ensure the exact firmware version (patch version starting with 0) is accessible for upgrade or downgrade.
   - Confirm the device supports installing older firmware if a downgrade will be performed.
2. Update or downgrade the firmware to a version where the patch version starts with 0.
   - The patch version refers to the third segment of the firmware version string (e.g., `1.2.01` → patch version is `01`).
   - Select a specific firmware version (newer or older) that meets this requirement.
3. Check if firmware-updater correctly displays the Current Version.

### Ubuntu release

24.04 LTS

### What architecture are you using?

amd64

d-loose, 2025-08-12:
Thanks for the report! I doubt that this is a UI issue, since the version is provided as a string by the backend. Could you please run `fwupdmgr get-updates` and see if the issue shows up in the CLI as well? If that's the case, please report the bug [upstream](https://github.com/fwupd/fwupd/issues).

d-loose, 2025-08-12:
Interesting, that definitely looks like a bug :/

Could you run `fwupdmgr get-devices` to find out the `Device ID` and `Current version` of the devices for which you see the issue? Afterwards you can get the list of available releases with `fwupd get-releases DEVICE-ID` - does the same inconsistency show up there?

d-loose, 2025-08-12:
Thanks a lot, yes looks like it! Would you mind opening an upstream issue?

Lidoca, 2025-08-12:
I mentioned the issue to fwupd. I don't think there's anything else to do here, right?

hughsie, 2025-08-12:
You should absolutely not use a string compare to see if an update is available. Each release has a flag to signify if it's an upgrade or downgrade, e.g. `FWUPD_RELEASE_FLAG_IS_UPGRADE` -- that'll work even with the cases where devices can't downgrade due to policy and stuff like that.

berkus, 2026-11-09:
Seems to be a bug in fwupd?

(fwupdmgr:515026): FuMain-DEBUG: 14:28:52.165: current version is 1.2.4: 1.2.04=same, 1.2.03=older

hughsie, 2026-11-09:
Why is it a bug?

    $ /usr/bin/fwupdtool  vercmp 1.2.4 1.2.04
    1.2.4 == 1.2.04

berkus, 2026-11-09:
> 15\. Why is it a bug?
>          ```
>          $ /usr/bin/fwupdtool  vercmp 1.2.4 1.2.04
>          1.2.4 == 1.2.04
>          ```

Because it displays 1.2.4 for 1.2.04 - this is not what the update specifies. It compares correctly, but the visual representation is incorrect.

In the UI it's visible as two different versions in "available to upgrade" vs "installed version": new update shows available version 1.2.04, and after installation it shows as 1.2.4.

hughsie, 2026-11-09:
That's really a vendor decision -- we could certainly "fix up" the device with the version from the metadata but I'm not sure that wouldn't break other tools

berkus, 2026-11-09:
> 17\. That's really a vendor decision -- we could certainly "fix up" the device with the version from the metadata but I'm not sure that wouldn't break other tools

I might be missing something, but my understanding is this: _metadata.xml defines how the version is displayed, so if it says 1.2.04 this is what we should display. We probably shouldn't touch the version that is encoded somewhere else (if there is such place - i do not know), but in this case we probably need to carry both the integer version and a string version view.

Regarding the fake device I found that fwupd test plugin always treats the version as integer, so it's impossible to test the "1.2.04" scenario, it will always be converted into 1.2.4 after [`fu_test_plugin_get_version()`](https://github.com/fwupd/fwupd/blob/2.1.6/plugins/test/fu-test-plugin.c#L147).

hughsie, 2026-11-09:
> but my understanding is this: _metadata.xml defines how the version is displayed

That's how the release gets displayed. If we don't have that exact release we use the normal "triplet" format. If we fix up the releases we do have on the LVFS and not the others (as we don't know the string value) we could have something like:

 * 1.2.05 (on LVFS)
 * 1.2.4 (not on LVFS)
 * 1.2.03 (on LVFS)

...which would be super weird. It's probably a better idea to "fixup" the metadata on the LVFS to match the triplet fwupd version. If you'd like to ponder that please open an issue and we can discuss there: https://gitlab.com/fwupd/lvfs-website/-/work_items

berkus, 2026-11-09:
> > but my understanding is this: _metadata.xml defines how the version is displayed
> 
> That's how the release gets displayed. 

model.latestRelease: 1.2.04 <- release, yes
device.version: 1.2.4 <- device, no

so now this same installed version doesn't display as current, either, because firmware-updater uses string comparison to compare these two versions.

With regards to this current ticket, it's a wontfix until we fix the display on the fwupd side to match what was installed - do you mean this kind of synchronization for the GitLab ticket?

hughsie, 2026-11-09:
> because firmware-updater uses string comparison to compare these two versions

I'm telling you that's probably a bad plan -- you need to look at the flag to decide if something is an upgrade or downgrade.

berkus, 2026-11-09:
> 21\. > because firmware-updater uses string comparison to compare these two versions
>          
>          
>          I'm telling you that's probably a bad plan -- you need to look at the flag to decide if something is an upgrade or downgrade.

Yes, that part is fixed. However, for consistency I would say it should show installed version as 1.2.04 if the pending installation version was displayed as 1.2.04. Just to lower the confusion levels.